gtfactslab/CrazySim #2: acados_ocp_solver_pyx

Kazimbalti, 2024-02-14:
Hello Sir,
  when I run "ros2 run crazyflie_mpc crazyflie_multiagent_mpc --n_agents=4"
  Please guide me how to resolve this issue, thank you
  
  File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1004, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'c_generated_code.acados_ocp_solver_pyx'
[ros2run]: Process exited with failure 1

llanesc, 2024-02-14:
Did you install Acados and run one of their Python examples to confirm it's properly set up? Also, have you sourced the ros2 workspace after building?

Kazimbalti, 2024-02-16:
Thank you @llanesc  the problem is solved,
but when I spawn more than 3 agents then after this command
"ros2 topic pub -t 50 -r 50 /all/mpc_takeoff std_msgs/msg/Empty"
the agents didn't stable,
some agents move high, some agents fall down and some agents return to ground again, as shown in Figure below
[image]

Please @llanesc guide me to resolve this issue, thank you

llanesc, 2024-02-17:
What is your Gazebo real time factor during the simulation? There is no lock stepping in the firmware, so when the Gazebo real time factor is too low, then the sensor data being sent to the firmware is coming in too slowly. You can reduce the real time factor by using a higher performance computer or using less drones. I'm considering implementing lock stepping in the future on this repo for running multi-agent simulation on lower performance computers.

closed 2024-04-06 by llanesc