gtfactslab/CrazySim #4: fatal error: lapack.h: No such file or directory

khunglong207, 2024-04-18:
[image]

i am having this problem can you help i tried the command line "sudo apt-get install liblapack-dev". But it still doesn't run

khunglong207, 2024-04-18:
i fixed it but i am having this problem
[image]

khunglong207, 2024-04-18:
I fixed it but when I go to connect I can't scan for udp, do you have any ideas?
[image]

khunglong207, 2024-04-18:
@llanesc I'm new so can you help me? I don't know that command

llanesc, 2024-04-18:
Right next to the address box in the client is a checkbox that says SITL. Click on that when you want to use the simulator.

khunglong207, 2024-04-18:
yes i clicked but not find udp

khunglong207, 2024-04-18:
[image]
@llanesc

llanesc, 2024-04-18:
You might have the official cflib installed. Can you try uninstalling cflib with pip uninstall cflib and then install it from the CrazySim folder with pip install -e . Also can you try running a [demo ](https://github.com/llanesc/crazyflie-lib-python/blob/master/examples/autonomy/motion_commander_demo.py) with CFLib to make sure it's installed. Just make sure to change the URI in the script to "udp://0.0.0.0:19850".

llanesc, 2024-04-18:
Also when you click on SITL checkbox you are clicking on scan and then connect right?

khunglong207, 2024-04-18:
theshy@theshy7119:~/CrazySim/crazyflie-lib-python$ pip install -e 

Usage:   
  pip install [options] <requirement specifier> [package-index-options] ...
  pip install [options] -r <requirements file> [package-index-options] ...
  pip install [options] [-e] <vcs project url> ...
  pip install [options] [-e] <local project path> ...
  pip install [options] <archive url/path> ...

-e option requires 1 argument
@llanesc I try but command show it i am a new student use ros drone for project so hope you try hepl me

khunglong207, 2024-04-18:
can you give me running a [demo ](https://github.com/llanesc/crazyflie-lib-python/blob/master/examples/autonomy/motion_commander_demo.py)with CFLib

llanesc, 2024-04-18:
go to the file: CrazySim/crazyflie-lib-python/examples/autonomy/motion_commander_demo.py and change the URI to uri_helper.uri_from_env(default='udp://0.0.0.0:19850')

khunglong207, 2024-04-18:
@llanesc oh my god thank you verry much i try it so much but when he answer me i do it. you r god.
I am a student aerospace in Hanoi university of science and technology In VỉetNam.
i have many question for your project but now In Hanoi is 1pm , it so late, so i will ask you for my question on tomorrow. Can i ask you from email? Once again I thank you, you are very enthusiastic in supporting me

llanesc, 2024-04-18:
Sure, just email me if you have anymore questions. christian.llanes@gatech.edu

closed 2024-04-18 by llanesc

GusBon, 2024-04-24:
> ![Screenshot from 2024-04-18 14-24-23](https://private-user-images.githubusercontent.com/142283805/323506063-fbeef7f8-1611-442e-8af3-1f5e7e2b2662.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTM5NjUyNzYsIm5iZiI6MTcxMzk2NDk3NiwicGF0aCI6Ii8xNDIyODM4MDUvMzIzNTA2MDYzLWZiZWVmN2Y4LTE2MTEtNDQyZS04YWYzLTFmNWU3ZTJiMjY2Mi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDI0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQyNFQxMzIyNTZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xMTA0MzYwNGVhMmQ0ZjFhZGQzMTU2ZjdlMDE4YWE0NTU4MTU3MDdkOWRkOTc1MGYxZmYwN2YzZDE0MjE0YWEzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.p670i4J9Fm4XwXqFCyEoSM5OB8sLFisI_mpBeGfiKrg)
> 
> i am having this problem can you help i tried the command line "sudo apt-get install liblapack-dev". But it still doesn't run

Hi,
Sorry to talk on this closed discussion, but I am facing the same issue with the lapack.h not found.
Could you explain how do you solve this issue @khunglong207 ?

llanesc, 2024-04-24:
@GusBon You need to install LAPACK.
Try doing this command:
```bash
sudo apt-get install liblapack-dev
```

GusBon, 2024-04-24:
Yup I have done it, as the dependencies of the section said to do, the package manager said that I already had liblapack-dev package. (I am running Ubuntu22.04)

llanesc, 2024-04-24:
@GusBon could you try this one as well?
```bash
sudo apt-get install liblapacke-dev
```

GusBon, 2024-04-25:
Hi,
It worked thanks !
After some researches liblapack is a fortran library for linear algebra.
the package "liblapack-dev" contains a static verions of the library,
and the package "liblapacke-dev" contains a static version of the library and development files for the C language for lapack.
I am now able to launch gazebo with a single crazyflie and 8 crazyflies in a square shape.
Thank you !