gtfactslab/CrazySim #5: colcon build --symlink-install error

khunglong207, 2024-04-18:
theshy@theshy7119:~/CrazySim$ cd ros2_ws
colcon build --symlink-install
Starting >>> crazyflie_interfaces
Starting >>> motion_capture_tracking_interfaces
Finished <<< motion_capture_tracking_interfaces [1.05s]                       
Starting >>> motion_capture_tracking
Finished <<< crazyflie_interfaces [1.73s]                                  
Starting >>> crazyflie_py
Starting >>> crazyflie                                                                          
Starting >>> crazyflie_sim
Finished <<< motion_capture_tracking [0.89s]                                                        
Finished <<< crazyflie [1.03s]                                                                   
--- stderr: crazyflie_sim                                                             
/home/theshy/.local/lib/python3.10/site-packages/setuptools/_distutils/cmd.py:66: SetuptoolsDeprecationWarning: setup.py install is deprecated.
!!

        ********************************************************************************
        Please avoid running ``setup.py`` directly.
        Instead, use pypa/build, pypa/installer or other
        standards-based tools.

        See https://blog.ganssle.io/articles/2021/10/setup-py-deprecated.html for details.
        ********************************************************************************

!!
  self.initialize_options()
---
Finished <<< crazyflie_sim [1.44s]
--- stderr: crazyflie_py                   
/home/theshy/.local/lib/python3.10/site-packages/setuptools/command/develop.py:40: EasyInstallDeprecationWarning: easy_install command is deprecated.
!!

        ********************************************************************************
        Please avoid running ``setup.py`` and ``easy_install``.
        Instead, use pypa/build, pypa/installer or other
        standards-based tools.

        See https://github.com/pypa/setuptools/issues/917 for details.
        ********************************************************************************

!!
  easy_install.initialize_options(self)
/home/theshy/.local/lib/python3.10/site-packages/setuptools/_distutils/cmd.py:66: SetuptoolsDeprecationWarning: setup.py install is deprecated.
!!

        ********************************************************************************
        Please avoid running ``setup.py`` directly.
        Instead, use pypa/build, pypa/installer or other
        standards-based tools.

        See https://blog.ganssle.io/articles/2021/10/setup-py-deprecated.html for details.
        ********************************************************************************

!!
  self.initialize_options()
---
Finished <<< crazyflie_py [2.15s]
Starting >>> crazyflie_examples
Starting >>> crazyflie_mpc                       
--- stderr: crazyflie_examples                                                              
/home/theshy/.local/lib/python3.10/site-packages/setuptools/_distutils/cmd.py:66: SetuptoolsDeprecationWarning: setup.py install is deprecated.
!!

        ********************************************************************************
        Please avoid running ``setup.py`` directly.
        Instead, use pypa/build, pypa/installer or other
        standards-based tools.

        See https://blog.ganssle.io/articles/2021/10/setup-py-deprecated.html for details.
        ********************************************************************************

!!
  self.initialize_options()
---
Finished <<< crazyflie_examples [1.09s]
--- stderr: crazyflie_mpc                   
/home/theshy/.local/lib/python3.10/site-packages/setuptools/command/develop.py:40: EasyInstallDeprecationWarning: easy_install command is deprecated.
!!

        ********************************************************************************
        Please avoid running ``setup.py`` and ``easy_install``.
        Instead, use pypa/build, pypa/installer or other
        standards-based tools.

        See https://github.com/pypa/setuptools/issues/917 for details.
        ********************************************************************************

!!
  easy_install.initialize_options(self)
/home/theshy/.local/lib/python3.10/site-packages/setuptools/_distutils/cmd.py:66: SetuptoolsDeprecationWarning: setup.py install is deprecated.
!!

        ********************************************************************************
        Please avoid running ``setup.py`` directly.
        Instead, use pypa/build, pypa/installer or other
        standards-based tools.

        See https://blog.ganssle.io/articles/2021/10/setup-py-deprecated.html for details.
        ********************************************************************************

!!
  self.initialize_options()
---
Finished <<< crazyflie_mpc [1.91s]

Summary: 8 packages finished [7.03s]
  4 packages had stderr output: crazyflie_examples crazyflie_mpc crazyflie_py crazyflie_sim
I'm trying to try the next steps of this pj but I'm getting an error in this part, can someone help me?

khunglong207, 2024-04-18:
and then i try Start up the Firmware in the folder ~/CrazySim/crazyflie-firmware
it work but then i try 
theshy@theshy7119:~/CrazySim/ros2_ws/src/crazyswarm2/crazyflie/launch$ ros2 launch crazyflie launch.py backend:=sim
command show :
Package 'crazyflie' not found: "package 'crazyflie' not found, searching: ['/opt/ros/humble']"
i think because colcon build --symlink-install fall

llanesc, 2024-04-18:
Did you run 
```bash
source install/setup.bash
```
from within ros2_ws folder? You need to do that every time you create a new terminal or you can put it in the ~/.bashrc folder to do it automatically. Take a look at the [ROS 2 tutorials](https://docs.ros.org/en/humble/Tutorials/Beginner-Client-Libraries/Creating-A-Workspace/Creating-A-Workspace.html#source-ros-2-environment) if you haven't yet.

khunglong207, 2024-04-18:
[image]
yeah but i dont see my drone in RViz 
@llanesc

llanesc, 2024-04-18:
so when you ran:
```bash
source install/setup.bash
```
and the error
Package 'crazyflie' not found: "package 'crazyflie' not found, searching: ['/opt/ros/humble']"
was gone? Can you show the terminal? Have you ran just 1 drone?

llanesc, 2024-04-18:
By default there will be 8 crazyflies enabled in this [config file](https://github.com/llanesc/crazyswarm2/blob/26d5d9829b087bbddd7abc94c2bae9ddeed0fbbd/crazyflie/config/crazyflies.yaml) you need to comment out the cf_5 through cf_8 as well or set the enabled flag to false. Also you have 4 crazyflies and the real time factor is already 62% which may be a problem when you want to run the MPC. You typically want 80%+ to get good performance. You'd need a computer with higher specs or try to run 1-2 drones.

khunglong207, 2024-04-18:
yes,the error
Package 'crazyflie' not found: "package 'crazyflie' not found, searching: ['/opt/ros/humble']"
was gone.
I ran 4 drone for your file readme example
[image]
[image]

llanesc, 2024-04-18:
Try running just 1 drone and comment out the cf_2 through cf_8 in this [config file](https://github.com/llanesc/crazyswarm2/blob/26d5d9829b087bbddd7abc94c2bae9ddeed0fbbd/crazyflie/config/crazyflies.yaml).

khunglong207, 2024-04-18:
> By default there will be 8 crazyflies enabled in this [config file](https://github.com/llanesc/crazyswarm2/blob/26d5d9829b087bbddd7abc94c2bae9ddeed0fbbd/crazyflie/config/crazyflies.yaml) you need to [...]

yeah, i turn off cf_5 through cf_8 is false from file yaml, i will try 1 or 2 drone, because my computer is not strong (i7 11800h, rtx 3060)

khunglong207, 2024-04-18:
@llanesc 
[image]
i try 1 drone but rviz no show drone

llanesc, 2024-04-18:
Can you show the other terminal? Looks like the drone is there. Zoom in because it's right under the world frame.

llanesc, 2024-04-18:
To test it out try running this in a different terminal.
```bash
ros2 service call /all/takeoff crazyflie_interfaces/srv/Takeoff \                       
"{duration: {sec: 2.0}, height: 1.0}"
```

khunglong207, 2024-04-18:
@llanesc 
the yaml file 
[image]
I took a look and saw that your file followed your script. Now I just need to type in your command to see your script, but I still can't see it. I don't know why. Maybe I'm a beginner use ros
[image]
[image]

khunglong207, 2024-04-18:
> To test it out try running this in a different terminal.
> 
> ```shell
> ros2 service call /all/takeoff crazyflie_interfaces/srv/Takeoff \                       
> "{duration: {sec: 2.0}, height: 1.0}"
> ```

@llanesc i try but it not work
[image]

khunglong207, 2024-04-18:
[image]
@llanesc  I seem to see cd_1 in world

llanesc, 2024-04-18:
you need to source the setup.bash file again because you opened a new terminal. put it in the bashrc file to not have to do that every time.

khunglong207, 2024-04-19:
> you need to source the setup.bash file again because you opened a new terminal. put it in the .bashrc file to not have to do that every time.
[image]

yeah, it works, amazinggg. you are my teacher. When can I reach your level? I admire it.  thank u so much. amz u r god
Currently I have 2 questions
Question 1: I want to fly like your spiral, what commands should I give accordingly?
Question 2: I'm trying to test the Start MPC code but I'm having some problems in the terminal. I'll take a photo in another issue.
terminal question2: 
theshy@theshy7119:~/CrazySim/ros2_ws$ ros2 run crazyflie_mpc crazyflie_multiagent_mpc --n_agents=1 --build_acados
Traceback (most recent call last):
  File "/home/theshy/CrazySim/ros2_ws/install/crazyflie_mpc/lib/crazyflie_mpc/crazyflie_multiagent_mpc", line 33, in <module>
    sys.exit(load_entry_point('crazyflie-mpc', 'console_scripts', 'crazyflie_multiagent_mpc')())
  File "/home/theshy/CrazySim/ros2_ws/install/crazyflie_mpc/lib/crazyflie_mpc/crazyflie_multiagent_mpc", line 25, in importlib_load_entry_point
    return next(matches).load()
  File "/usr/lib/python3.10/importlib/metadata/__init__.py", line 171, in load
    module = import_module(match.group('module'))
  File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/theshy/CrazySim/ros2_ws/build/crazyflie_mpc/crazyflie_mpc/crazyflie_multiagent_mpc.py", line 11, in <module>
    from .trajectory_tracking_mpc import TrajectoryTrackingMpc
  File "/home/theshy/CrazySim/ros2_ws/build/crazyflie_mpc/crazyflie_mpc/trajectory_tracking_mpc.py", line 1, in <module>
    from acados_template import AcadosOcp, AcadosOcpSolver, AcadosModel
ModuleNotFoundError: No module named 'acados_template'

llanesc, 2024-04-19:
you need to build acados from source and install the python interface in a different folder. https://docs.acados.org/installation/

khunglong207, 2024-04-19:
> you need to build acados from source and install the python interface in a different folder. https://docs.acados.org/installation/

yes i installed from this link but i try acados --version it not find acaddos

llanesc, 2024-04-19:
did you also install the python interface as well? https://docs.acados.org/python_interface/index.html

khunglong207, 2024-04-19:
> did you also install the python interface as well? Don't worry about the virtual env. https://docs.acados.org/python_interface/index.html

oh i don't install python_interface. 
theshy@theshy7119:~$ virtualenv env --python=/usr/bin/python3
created virtual environment CPython3.10.12.final.0-64 in 431ms
  creator CPython3Posix(dest=/home/theshy/env, clear=False, no_vcs_ignore=False, global=False)
  seeder FromAppData(download=False, pip=bundle, setuptools=bundle, wheel=bundle, via=copy, app_data_dir=/home/theshy/.local/share/virtualenv)
    added seed packages: pip==24.0, setuptools==59.6.0, wheel==0.43.0
  activators BashActivator,CShellActivator,FishActivator,NushellActivator,PowerShellActivator,PythonActivator
theshy@theshy7119:~$ source env/bin/activate
bash: env/bin/activate: No such file or directory
theshy@theshy7119:~$ 
this command when i install

llanesc, 2024-04-19:
you can skip step 2 and go to 3.

khunglong207, 2024-04-19:
bash: acados_root: No such file or directory
but command show

llanesc, 2024-04-19:
replace the <acados_root> with the absolute path of your acados folder.

khunglong207, 2024-04-19:
pip install -e /home/theshy/acados/interfaces/acados_template
yeah i install in acados

khunglong207, 2024-04-19:
[image]
i installed from your link, so next step what can i do

llanesc, 2024-04-19:
Put this in the bashrc.
```bash
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:"/home/theshy/acados/lib"
export ACADOS_SOURCE_DIR="/home/theshy/acados/
```
Then try running this in a new terminal after you have put the last export variables in the bashrc.
```bash
ros2 run crazyflie_mpc crazyflie_multiagent_mpc --n_agents=1 --build_acados
```

khunglong207, 2024-04-19:
theshy@theshy7119:~/CrazySim/ros2_ws$ ros2 run crazyflie_mpc crazyflie_multiagent_mpc --n_agents=1 --build_acados
Acados cython code not generated. Generating cython code now...
Warning: Did not find environment variable ACADOS_SOURCE_DIR, guessed ACADOS_PATH to be /home/theshy/acados.
Please export ACADOS_SOURCE_DIR to avoid this warning.
Warning: Please note that the following versions of CasADi are officially supported: 3.4.03.4.5 or 3.5.1 or 3.5.2 or 3.5.3 or 3.5.4 or 3.5.6 or 3.5.5 or 3.6.0 or 3.6.1 or 3.6.2 or 3.6.3 or 3.6.4.
 If there is an incompatibility with the CasADi generated code, please consider changing your CasADi version.
Version 3.6.5 currently in use.
Warning: Did not find environment variable ACADOS_SOURCE_DIR, guessed ACADOS_PATH to be /home/theshy/acados.
Please export ACADOS_SOURCE_DIR to avoid this warning.
Warning: Did not find environment variable ACADOS_SOURCE_DIR, guessed ACADOS_PATH to be /home/theshy/acados.
Please export ACADOS_SOURCE_DIR to avoid this warning.
Warning: Did not find environment variable ACADOS_SOURCE_DIR, guessed ACADOS_PATH to be /home/theshy/acados.
Please export ACADOS_SOURCE_DIR to avoid this warning.
Warning: Did not find environment variable ACADOS_SOURCE_DIR, guessed ACADOS_PATH to be /home/theshy/acados.
Please export ACADOS_SOURCE_DIR to avoid this warning.
Warning: Did not find environment variable ACADOS_SOURCE_DIR, guessed ACADOS_PATH to be /home/theshy/acados.
Please export ACADOS_SOURCE_DIR to avoid this warning.
Warning: Did not find environment variable ACADOS_SOURCE_DIR, guessed ACADOS_PATH to be /home/theshy/acados.
Please export ACADOS_SOURCE_DIR to avoid this warning.
Warning: Did not find environment variable ACADOS_SOURCE_DIR, guessed ACADOS_PATH to be /home/theshy/acados.
Please export ACADOS_SOURCE_DIR to avoid this warning.
Warning: Did not find environment variable ACADOS_SOURCE_DIR, guessed ACADOS_PATH to be /home/theshy/acados.
Please export ACADOS_SOURCE_DIR to avoid this warning.
Warning: Did not find environment variable ACADOS_SOURCE_DIR, guessed ACADOS_PATH to be /home/theshy/acados.
Please export ACADOS_SOURCE_DIR to avoid this warning.
Warning: Did not find environment variable ACADOS_SOURCE_DIR, guessed ACADOS_PATH to be /home/theshy/acados.
Please export ACADOS_SOURCE_DIR to avoid this warning.
Warning: Did not find environment variable ACADOS_SOURCE_DIR, guessed ACADOS_PATH to be /home/theshy/acados.
Please export ACADOS_SOURCE_DIR to avoid this warning.
Warning: Did not find environment variable ACADOS_SOURCE_DIR, guessed ACADOS_PATH to be /home/theshy/acados.
Please export ACADOS_SOURCE_DIR to avoid this warning.
Warning: Did not find environment variable ACADOS_SOURCE_DIR, guessed ACADOS_PATH to be /home/theshy/acados.
Please export ACADOS_SOURCE_DIR to avoid this warning.
Warning: Did not find environment variable ACADOS_SOURCE_DIR, guessed ACADOS_PATH to be /home/theshy/acados.
Please export ACADOS_SOURCE_DIR to avoid this warning.
Warning: Did not find environment variable ACADOS_SOURCE_DIR, guessed ACADOS_PATH to be /home/theshy/acados.
Please export ACADOS_SOURCE_DIR to avoid this warning.
Warning: Did not find environment variable ACADOS_SOURCE_DIR, guessed ACADOS_PATH to be /home/theshy/acados.
Please export ACADOS_SOURCE_DIR to avoid this warning.
Warning: Did not find environment variable ACADOS_SOURCE_DIR, guessed ACADOS_PATH to be /home/theshy/acados.
Please export ACADOS_SOURCE_DIR to avoid this warning.
Warning: Did not find environment variable ACADOS_SOURCE_DIR, guessed ACADOS_PATH to be /home/theshy/acados.
Please export ACADOS_SOURCE_DIR to avoid this warning.
make: *** No rule to make target 'clean_all'.  Stop.
cc -fPIC -std=c99   -O2 -I/home/theshy/acados/include -I/home/theshy/acados/include/acados -I/home/theshy/acados/include/blasfeo/include -I/home/theshy/acados/include/hpipm/include  -c -o acados_solver_crazyflie.o acados_solver_crazyflie.c
cc -fPIC -std=c99   -O2 -I/home/theshy/acados/include -I/home/theshy/acados/include/acados -I/home/theshy/acados/include/blasfeo/include -I/home/theshy/acados/include/hpipm/include  -c -o crazyflie_model/crazyflie_expl_ode_fun.o crazyflie_model/crazyflie_expl_ode_fun.c
cc -fPIC -std=c99   -O2 -I/home/theshy/acados/include -I/home/theshy/acados/include/acados -I/home/theshy/acados/include/blasfeo/include -I/home/theshy/acados/include/hpipm/include  -c -o crazyflie_model/crazyflie_expl_vde_forw.o crazyflie_model/crazyflie_expl_vde_forw.c
cc -fPIC -std=c99   -O2 -I/home/theshy/acados/include -I/home/theshy/acados/include/acados -I/home/theshy/acados/include/blasfeo/include -I/home/theshy/acados/include/hpipm/include  -c -o crazyflie_model/crazyflie_expl_vde_adj.o crazyflie_model/crazyflie_expl_vde_adj.c
cc -shared acados_solver_crazyflie.o crazyflie_model/crazyflie_expl_ode_fun.o crazyflie_model/crazyflie_expl_vde_forw.o crazyflie_model/crazyflie_expl_vde_adj.o -o libacados_ocp_solver_crazyflie.so -L/home/theshy/acados/lib -lacados -lhpipm -lblasfeo -lm \
-L -l
cython \
-o acados_ocp_solver_pyx.c \
-I /home/theshy/acados/include/../interfaces/acados_template/acados_template \
/home/theshy/acados/include/../interfaces/acados_template/acados_template/acados_ocp_solver_pyx.pyx \
-I /home/theshy/CrazySim/ros2_ws/install/crazyflie_mpc/share/crazyflie_mpc/acados_generated_files/c_generated_code \

/usr/lib/python3/dist-packages/pythran/tables.py:4520: FutureWarning: In the future `np.bool` will be defined as the corresponding NumPy scalar.
[... cut ...]
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 674, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 571, in module_from_spec
  File "<frozen importlib._bootstrap_external>", line 1176, in create_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
ImportError: libhpipm.so: cannot open shared object file: No such file or directory
[ros2run]: Process exited with failure 1
@llanesc this command show

llanesc, 2024-04-19:
you didn't export the variables correctly. can you show the bashrc file?

khunglong207, 2024-04-19:
> n crazyflie_mpc crazyflie_multiagent_mpc --n_agents=1 --build_acados

[image]
this bashrc file

llanesc, 2024-04-19:
try restarting the terminal. the acados lib path isn't in the variable.

khunglong207, 2024-04-19:
> try restarting the terminal. the acados lib path isn't in the variable.

theshy@theshy7119:~/CrazySim/ros2_ws$ ros2 run crazyflie_mpc crazyflie_multiagent_mpc --n_agents=1 --build_acados
Acados cython module imported successfully.
Warning: Please note that the following versions of CasADi are officially supported: 3.4.03.4.5 or 3.5.1 or 3.5.2 or 3.5.3 or 3.5.4 or 3.5.6 or 3.5.5 or 3.6.0 or 3.6.1 or 3.6.2 or 3.6.3 or 3.6.4.
 If there is an incompatibility with the CasADi generated code, please consider changing your CasADi version.
Version 3.6.5 currently in use.
make: *** No rule to make target 'clean_all'.  Stop.
cc -fPIC -std=c99   -O2 -I/home/theshy/acados/include -I/home/theshy/acados/include/acados -I/home/theshy/acados/include/blasfeo/include -I/home/theshy/acados/include/hpipm/include  -c -o acados_solver_crazyflie.o acados_solver_crazyflie.c
cc -fPIC -std=c99   -O2 -I/home/theshy/acados/include -I/home/theshy/acados/include/acados -I/home/theshy/acados/include/blasfeo/include -I/home/theshy/acados/include/hpipm/include  -c -o crazyflie_model/crazyflie_expl_ode_fun.o crazyflie_model/crazyflie_expl_ode_fun.c
cc -fPIC -std=c99   -O2 -I/home/theshy/acados/include -I/home/theshy/acados/include/acados -I/home/theshy/acados/include/blasfeo/include -I/home/theshy/acados/include/hpipm/include  -c -o crazyflie_model/crazyflie_expl_vde_forw.o crazyflie_model/crazyflie_expl_vde_forw.c
cc -fPIC -std=c99   -O2 -I/home/theshy/acados/include -I/home/theshy/acados/include/acados -I/home/theshy/acados/include/blasfeo/include -I/home/theshy/acados/include/hpipm/include  -c -o crazyflie_model/crazyflie_expl_vde_adj.o crazyflie_model/crazyflie_expl_vde_adj.c
cc -shared acados_solver_crazyflie.o crazyflie_model/crazyflie_expl_ode_fun.o crazyflie_model/crazyflie_expl_vde_forw.o crazyflie_model/crazyflie_expl_vde_adj.o -o libacados_ocp_solver_crazyflie.so -L/home/theshy/acados/lib -lacados -lhpipm -lblasfeo -lm \
-L -l
cython \
-o acados_ocp_solver_pyx.c \
-I /home/theshy/acados/include/../interfaces/acados_template/acados_template \
/home/theshy/acados/include/../interfaces/acados_template/acados_template/acados_ocp_solver_pyx.pyx \
-I /home/theshy/CrazySim/ros2_ws/install/crazyflie_mpc/share/crazyflie_mpc/acados_generated_files/c_generated_code \

/usr/lib/python3/dist-packages/pythran/tables.py:4520: FutureWarning: In the future `np.bool` will be defined as the corresponding NumPy scalar.
  if not hasattr(numpy, method):
/usr/lib/python3/dist-packages/pythran/tables.py:4553: FutureWarning: In the future `np.bytes` will be defined as the corresponding NumPy scalar.
  obj = getattr(themodule, elem)
cc  -c -O2 \
-fPIC \
-o acados_ocp_solver_pyx.o \
-I /home/theshy/acados/include/blasfeo/include/ \
-I /home/theshy/acados/include/hpipm/include/ \
-I /home/theshy/acados/include \
-I /home/theshy/.local/lib/python3.10/site-packages/numpy/core/include \
-I /usr/include/python3.10 \
acados_ocp_solver_pyx.c \

In file included from /home/theshy/.local/lib/python3.10/site-packages/numpy/core/include/numpy/ndarraytypes.h:1940,
                 from /home/theshy/.local/lib/python3.10/site-packages/numpy/core/include/numpy/ndarrayobject.h:12,
                 from /home/theshy/.local/lib/python3.10/site-packages/numpy/core/include/numpy/arrayobject.h:5,
                 from acados_ocp_solver_pyx.c:1230:
/home/theshy/.local/lib/python3.10/site-packages/numpy/core/include/numpy/npy_1_7_deprecated_api.h:17:2: warning: #warning "Using deprecated NumPy API, disable it with " "#define NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION" [-Wcpp]
   17 | #warning "Using deprecated NumPy API, disable it with " \
      |  ^~~~~~~
cc  -shared \
-o acados_ocp_solver_pyx.so \
-Wl,-rpath=/home/theshy/acados/lib \
acados_ocp_solver_pyx.o \
/home/theshy/CrazySim/ros2_ws/install/crazyflie_mpc/share/crazyflie_mpc/acados_generated_files/c_generated_code/libacados_ocp_solver_crazyflie.so \
-L/home/theshy/acados/lib -lacados -lhpipm -lblasfeo -lm
Acados cython module imported successfully.
[INFO] [1713487435.377698331] [cf_1]: Initialization completed...
[INFO] [1713487435.394704045] [cf_1]: Beginning multiagent executor, shut down with CTRL-C
it works! thanks u so much. So what is next step

llanesc, 2024-04-19:
Run this takeoff command
```bash
ros2 topic pub -t 1 /all/mpc_takeoff std_msgs/msg/Empty
```
then command MPC to follow the trajectory
```bash
ros2 topic pub -t 1 /all/mpc_trajectory std_msgs/msg/Empty
```

khunglong207, 2024-04-19:
> Run this takeoff command
> 
> ```shell
> ros2 topic pub -t 1 /all/mpc_takeoff std_msgs/msg/Empty
> ```
> 
> then command MPC to follow the trajectory
> 
> ```shell
> ros2 topic pub -t 1 /all/mpc_trajectory std_msgs/msg/Empty
> ```

[image]
Looks like it's not running

llanesc, 2024-04-19:
In a different terminal. run the takeoff command first and wait for it to hover in place before running the next command.

khunglong207, 2024-04-19:
> In a different terminal. run the takeoff command first and wait for it to hover in place before running the next command.

theshy@theshy7119:~/CrazySim/ros2_ws$ ros2 service call /all/takeoff crazyflie_interfaces/srv/Takeoff \                       
"{duration: {sec: 2.0}, height: 1.0}"
Failed to populate field: Value 'None' is expected to be a dictionary but is a NoneType
{duration: {sec: 2.0}, height: 1.0}: command not found
i get fail when i take off your drone

khunglong207, 2024-04-19:
[image]
sr i try it but i get faild in command

llanesc, 2024-04-19:
1. delete that terminal
2. open a new terminal
3. run ros2 run crazyflie_mpc crazyflie_multiagent_mpc --n_agents=1 --build_acados
4. open a new terminal
5. ros2 topic pub -t 1 /all/mpc_takeoff std_msgs/msg/Empty

khunglong207, 2024-04-19:
> 1. delete that terminal
> 2. open a new terminal
> 3. run ros2 run crazyflie_mpc crazyflie_multiagent_mpc --n_agents=1 --build_acados
> 4. open a new terminal

[image]
@llanesc it not works, teacher

llanesc, 2024-04-19:
The crazyflie_mpc is not receiving the state messages correctly from the firmware. Close everything and try restarting gazebo, crazyswarm2, and the multiagent_mpc node. Sometimes this happens if the packet was lost requesting the pose state information.

khunglong207, 2024-04-19:
> The crazyflie_mpc is not receiving the state messages correctly from the firmware. Close everything and try restarting gazebo, crazyswarm2, and the multiagent_mpc node. Sometimes this happens if the packet was lost requesting the pose state information.

[image]
but i try ros2 topic pub -t 1 /all/mpc_trajectory std_msgs/msg/Empty it not change

llanesc, 2024-04-19:
seems like you send the takeoff command twice try the trajectory one

khunglong207, 2024-04-19:
> seems like you sent the takeoff command twice try the trajectory one

@llanesc yes sir, my teach i did drone fly as image your git , If I want to program to change the trajectory instead of a spiral (for example a square), what steps do I need and how will the trajectory file be configured?

llanesc, 2024-04-19:
You would need to change the variable self.trajectory_type in this [file](https://github.com/llanesc/crazyflie_mpc/blob/e37eae23f99f8edfb303f1e3c2dd99685de5e9ca/crazyflie_mpc/crazyflie_multiagent_mpc.py). There's a few available already in the trajectory_function().

khunglong207, 2024-04-19:
@llanesc I'm actually very grateful to you, because you've helped me so much because I'm a college student so ros is something very new to me so I have a lot of questions (I think it will bother you). but you still answered me fully and very easily) you are not only good but you are very kind. I am very grateful to you. I hope you will support me when I encounter some problems in this pj. Once again I thank you very much. Wishing you good health and many more great projects my teacher

llanesc, 2024-04-19:
Happy to help!

closed 2024-04-19 by llanesc