gtfactslab/CrazySim #6: Unreliable Initialization of Crazyflies

KhAlamdar11, 2024-04-28:
Hello,

I am encountering an issue when launching the multiagent Crazyflie simulation with the following commands:

1. `bash tools/crazyflie-simulation/simulator_files/gazebo/launch/sitl_multiagent_square.sh -n 4 -m crazyflie`
2. `ros2 launch crazyflie launch.py backend:=cflib`

**Expected behaviour:**
`...`
`[crazyflie_server.py-4] [INFO] [1714264937.928823133] [crazyflie_server]: All Crazyflies loggging are initialized`

**Current Behavior:**
Does not initialize all UAVs, and stops after connecting to a certain cf. It is random; sometimes it stops after connecting to 2, sometimes 3 and so on.
`...`
`[crazyflie_server.py-4] [INFO] [1714259004.921884676] [crazyflie_server]:  udp://0.0.0.0:19854 is fully connected!`

Additionally, does not publishes topics related to cfs so you cannot see the pose in rviz of any cf either. It appears to work correctly approximately once in every ten attempts. I have attached log files of both success and failure cases.
[success.log](https://github.com/gtfactslab/CrazySim/files/15140406/success.log)
[failure.log](https://github.com/gtfactslab/CrazySim/files/15140408/failure.log)

**Environment**
Recommended environment is used: Ubuntu 22.04, ROS2 Humble and Gazebo Garden.
To troubleshoot, I've confirmed that the number of Crazyflies specified in the simulation matches those in the Crazyflie configuration file. I've also tried delaying the second command after launching gazebo, but there's no consistent pattern to when the initialization is successful.

llanesc, 2024-04-28:
I've encountered a similar problem with larger swarm sizes (greater than 20). I suspected this had to do with some packet losses in the socket or timeouts occurring on CFLib. What is your Gazebo real time factor when all 5 drones are running and just before launching CrazySwarm2? Have you encountered this issue with 1-2 drones on your machine?

KhAlamdar11, 2024-04-28:
The issue was related to the Gazebo real-time factor as you highlighted. With larger swarms, the real-time factor decreased, increasing the likelihood of connection failures. Following your recommendation from a previous issue, I increased the Gazebo step size. This adjustment helped maintain the real-time factor close to 1, ensuring successful connections every time.
Thank you for the help!

closed 2024-04-28 by KhAlamdar11