gtfactslab/CrazySim #7: problem with build and lapack library

matheeusper, 2024-04-30:
[image]
i'm trying run your project SITL. But doesn't work, and i don't know how fix, did you has how can i fix that?

DexterZzz1010, 2024-04-30:
@matheeusper  try this
`sudo apt-get install liblapacke-dev`

matheeusper, 2024-04-30:
work! but now i have the error:
[image]

DexterZzz1010, 2024-04-30:
Maybe you also have to install opencv. But I didn't meet this error before.

llanesc, 2024-05-07:
Going to close this. Should be fixed in e6afa2941a248129cd7668af45b99f28af8870b3 after removing unnecessary dependencies.

closed 2024-05-07 by llanesc