gtfactslab/CrazySim #9: Modified version of the CFlib backend and cmd_vel_legacy topic

GusBon, 2024-05-14:
Hi,
I am trying to control a Crazyflie by sending to it roll pitch yaw and thrust commands. I have developed a node that publish on topics of Crazyswarm2.

But after some tests, the node appears to be working but nothing happens on Crazysim, the propellers of the Crazyflie are not turning. I am publishing on the cmd_vel_legacy topic. The node is publishing correctly on the topic when I echo it and the crazyflie servers seems to receive the data also.
For more details, I have already discussed about it on the Crazyswarm2 github (https://github.com/IMRCLab/crazyswarm2/discussions/499).

I have tested with another node to publish on the cmd_hover topic, and this time everything works fine, the drone fly to the wanted position.

llanesc, 2024-05-15:
It could be a few things. I am using a slightly modified version of Crazyswarm2 which is a bit outdated. The modifications I made are to directly control attitude setpoints through its own [topic](https://github.com/llanesc/crazyswarm2/blob/26d5d9829b087bbddd7abc94c2bae9ddeed0fbbd/crazyflie/scripts/crazyflie_server.py#L245) instead of the cmd_vel_legacy option. Although that shouldn't stop you from using cmd_vel_legacy. Are you initially sending all zeros like I do [here](https://github.com/llanesc/crazyflie_mpc/blob/e37eae23f99f8edfb303f1e3c2dd99685de5e9ca/crazyflie_mpc/crazyflie_multiagent_mpc.py#L294). I believe this a requirement from the firmware when you want to switch to the low level commander. Otherwise the firmware will reject all the inputs and the motors won't spin.

GusBon, 2024-05-16:
Ok,
Thank for this answer and for the precision on the attitude set points topic.
So, has you said the problem was that i didn't publish all zeros at the beginning of the sim.
It is now working by doing so.
Thanks a lot ! : )

closed 2024-05-16 by GusBon

enessas, 2025-02-26:
I am sending all zeros, but it doesn't work for me. What am I doing wrong?

When I send zeros to /cmd_attitude_setpoint and give a command, it works fine. But I couldn't achieve the same for /cmd_vel_legacy.

For /cmd_attitude_setpoint, this command works as expected:

ros2 topic pub --once /cf_2/cmd_attitude_setpoint crazyflie_interfaces/msg/AttitudeSetpoint '{header: {stamp: {sec: 0, nanosec: 0}, frame_id: ""}, roll: 0.0, pitch: 0.0, yaw_rate: 0.0, thrust: 0}'

However, when I try the following for /cmd_vel_legacy, it doesn't have any effect:

ros2 topic pub --once /cf_2/cmd_vel_legacy geometry_msgs/msg/Twist '{linear: {x: 0.0, y: 0.0, z: 0.0}, angular: {x: 0.0, y: 0.0, z: 0.0}}'

What am I doing wrong?