gtfactslab/CrazySim #10: Crazyswarm2 launch error

DexterZzz1010, 2024-05-19:
I'm trying to use a real drone and have modified the corresponding configuration files, but I can't start the launch file.

Also, I've noticed that this launch file is somewhat different from the main branch in the official repo.

 I would like to ask if the MPC controller can be used on a real drone, and how exactly should it be operated?

Error:
`ros2 launch crazyflie launch.py
[INFO] [launch]: All log files can be found below /home/zyf/.ros/log/2024-05-19-19-55-13-826536-zyf-ASUS-15585
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): Caught multiple exceptions when trying to load file of format [py]:
 - TypeError: 'NoneType' object is not subscriptable
 - InvalidFrontendLaunchFileError: The launch file may have a syntax error, or its format is unknown
`

llanesc, 2024-05-22:
Hi @DexterZzz1010, you can definitely use the MPC controller on a real drone. You will need to change the crazyswarm2 config file so that the robot type is cf instead of cf_sim along with the uri for the radio. You will need to change the motion capture system as well to whatever system you have. I don't have the flow deck or the loco deck, so I haven't tried anything other than motion capture cameras. It's probably possible to use by just deactivating the motion capture in the config file. Also you will need to use the following launch command with the cflib backend:
```bash
ros2 launch crazyflie launch.py backend:=cflib
```
My Crazyswarm2 repo is a few months out of date, but I can update it to the latest main when I get back home from international travel.

DexterZzz1010, 2024-05-28:
Hi！
Sorry for the late reply. I've updated Crazyswarm2 and the motion capture, and excitingly, there are no issues with ROS communication; all topics are successfully subscribed.
However, when using the MPC, the solver encountered an error. We believe this is due to inaccuracies in the Lighthouse positioning system. We introduced significant noise in the simulation system and reduced the constraints, which resulted in the same error. Currently, we don't have a good solution to this problem. 
But still, thank you for your detailed response.

closed 2024-05-28 by DexterZzz1010

llanesc, 2024-05-28:
I haven't tested the robustness of the MPC controller to significant noise before, so I haven't encountered this issue. I'll think about this problem a bit more and try to get back to you. Hopefully you are able to resolve it though.