gtfactslab/CrazySim #12: Problems with crazyflie-clients-python

johnstonk9, 2024-05-31:
For some reason cfclient doesn't want to work for me. I've tried cloning the "git clone https://github.com/llanesc/crazyflie-clients-python" both in my home directory and the Crazysim directory, but neither want to work. I have had uninstalled and reinstalled the package multiple times and that could be the problem but I am not sure.

Here was an installation attempt:
[image]

And this was the error message:
[image]

I try changing the main.py because I saw that the code went from .setValue to .setText and thought that adding str(address) would fix it, and the client launched but even in SITL mode I couldn't connect:
[image]

[image]

Do you know if the issue is with how I installed it and dependencies, or if there is some other error? Thanks.

llanesc, 2024-05-31:
I pushed a [commit](https://github.com/llanesc/crazyflie-clients-python/commit/dd9ba1941ed15cdeebe288aee3810e3db60c74c6) that seems to have fixed this. The problem is that I had to create a new AddressBox class to switch between hex and ip addresses. I wasn't getting the error, but I decided to just remove it and allow the SITL check to be done when connecting and detecting the link is udp vs radio or usb. Pull the latest sitl-release branch and check if it works now. If you encounter other issues let me know.

llanesc, 2024-06-06:
Hi @johnstonk9, were you able to test if the [commit](https://github.com/llanesc/crazyflie-clients-python/commit/dd9ba1941ed15cdeebe288aee3810e3db60c74c6) fixed your issue?

closed 2024-06-13 by llanesc

johnstonk9, 2024-06-13:
looks like everything works now! I deleted everything and reinstalled them just to make sure so not sure if that's the reason it works or if there was an actual issue with cfclient.

llanesc, 2024-06-13:
@johnstonk9 There was an issue I wasn't aware of because it wasn't showing up on my desktop machine. I did get the error on my laptop and it should be fixed now.