gtfactslab/CrazySim #14: how to add custom firmware

hyeminsong, 2024-06-20:
Hello!

I want to use this project for checking my firmware.

```
mkdir -p sitl_make/build && cd $_
cmake ..
make all
```

I did the build above and also checked that /sitl_make/CMakeList.txt has a list of codes that I added to /src. 
(${CF2_SITL_SRCS_DIR}/modules/src/controller/controller..c)
I want to add my controller, is there anything else that needs to be done separately?

Additionally, I was wondering if you could know how to add multiranger deck in simulation(gazebo).

Thank you!

llanesc, 2024-06-20:
Hi @hyeminsong, you can follow this [blog post](https://www.bitcraze.io/2023/02/adding-an-estimator-or-controller/) for steps on how to add a new controller. The steps to add to KBuild can be replaced by the CMakeLists.txt. I plan to switch over to KBuild soon to be consistent with the official firmware. Then to use your controller you need to modify the [stabilizer.controller](https://www.bitcraze.io/documentation/repository/crazyflie-firmware/master/api/params/#stabilizercontroller) parameter through CFLib, crazyflie-client, Crazyswarm2, etc. Right now I don't currently support the multiranger deck. I think this will require setting up 5 linear versions of the [lidar sensor](https://gazebosim.org/docs/garden/sensors#lidar-sensor) with different poses on the cf base_link. This can be done in the [models folder](https://github.com/llanesc/crazyflie-simulation/tree/61dd124b8560807c2762b1aa3f2c142e91973edb/simulator_files/gazebo/models) by creating a new model called crazyflie_multiranger. Then, you will need a new plugin or modify the old one to subscribe to those 5 lidar messages and send it to the firmware using a custom CRTP packet like what was done for the IMU. The issue is data transmission to the firmware. The deck code is typically heavily embedded in the firmware with device comms, so it may require a custom file based on this [range source code](https://github.com/bitcraze/crazyflie-firmware/blob/master/src/modules/src/range.c#L37) that logs to the [range](https://www.bitcraze.io/documentation/repository/crazyflie-firmware/master/api/logs/#range) log group.

hyeminsong, 2024-06-21:
Thank you for your kind reply.  I think I didn't explain my situation properly, so I'm adding content.

I've already tested it in hardware with custom controller. I'd like to proceed with further modification on the simulation with your project. Currently, I can't receive the data of the log group I added to controller at this time(TOC Not exist error). So, I think this error came form controller not added properly.  Because of that, I wonder if additional work is needed in the simulation.

llanesc, 2024-06-21:
So you tested your controller on hardware and you can see the custom log group on the client/cflib, but when you moved it to the CrazySim firmware you cannot? There shouldn't be any extra steps other than adding the source files to the CMakeLists file in sitl_make and registering the controller in the controller.c and controller.h file. Can you check if the log group exists by using the custom crazyflie-client and connecting to CrazySim?

closed 2024-11-14 by llanesc