gtfactslab/CrazySim #16: inconsistant connections

EdSolidworks, 2024-07-26:
I am having some issues with connecting to the simulator. When I try to use the cfclient to connect, it connects successfully about 20% of the time, otherwise, it freezes and I restart. The times that it connects successfully, I try to take off and it usually freezes. I have had the same issues with running high-level commander codes. It does work very well when it doesn't freeze.
I really need help with this and don't really know what to do, thank you anyone for any help.

I am running the software on Ubuntu 22.04, on oracle VM virtualbox. I have 16Gb of ram.

llanesc, 2024-07-26:
For reliable communication between Gazebo, crazyflie-firmware, and external scripts/client you need to have a Gazebo real time factor above like 70-80%. It's likely because you're using a virtual machine. I have never tested this system on a VM. What is your Gazebo real time factor? Additionally, see https://github.com/gtfactslab/CrazySim/issues/8 and https://github.com/gtfactslab/CrazySim/issues/1.

EdSolidworks, 2024-07-26:
It usually hovers around 70%. That seems to be the issue, thank you so much. It is working when the real time factor is about 75%.

closed 2024-07-26 by llanesc