gtfactslab/CrazySim #18: Simulation issue

ArkhamKnightNotRedHood, 2024-10-30:
Hello!
I followed the installation instructions to the point where I start the simulation via
`bash tools/crazyflie-simulation/simulator_files/gazebo/launch/sitl_singleagent.sh -m crazyflie -x 0 -y 0`
Then, I start cfclient, click on Scan, but I can't find the address udp://0.0.0.0:19850. 
And if I skip the step of cfclient, after all the installation finished, neither can I connect to the drone on gazebo simulation when I begin to run the cflib via
`ros2 launch crazyflie launch.py backend:=cflib`
So I realize that the issue may happened in the simulation step.  It’s possible that the simulated drone created is not connected to the address udp://0.0.0.0:19850? But what could have caused that?
By the way, In the logs of the simulation, there is a libEGL warning:
`libEGL warning: egl: failed to create dri2 screen`
Could it be caused by that? Regardless of whether it is or not, how should I resolve this issue?

llanesc, 2024-11-05:
Are you using the custom cflib package provided and custom cfclient (https://github.com/llanesc/crazyflie-clients-python)? You make have to remove them from pip and reinstall.

ArkhamKnightNotRedHood, 2024-11-08:
Thank you for your previous response. I have now completed the installation of CrazySim, but a new issue has arisen. When running CrazySwarm2, I found that the drone can only take off and land, but cannot move on the horizontal plane. Neither the goto() nor  startTrajectory() functions seem to work. Do you know what might be causing this issue?

> Are you using the custom cflib package provided and custom cfclient (https://github.com/llanesc/crazyflie-clients-python)? You may have to remove them from pip and reinstall.

ArkhamKnightNotRedHood, 2024-11-08:
> Are you using the custom cflib package provided and custom cfclient (https://github.com/llanesc/crazyflie-clients-python)? You may have to remove them from pip and reinstall.

Thank you for your previous response. I have now completed the installation of CrazySim, but a new issue has arisen. When running CrazySwarm2, I found that the drone can only take off and land normally, but cannot move normally on the horizontal plane. When I run goto(), All the drones are circling along the same trajectory, and then lose control after a period of time. When I run startTrajectory(), they just become completely motionless. And the log for it shows "Notify trajectory not yet implemented" and "Start trajectory not yet implemented", I'm not sure if it's normal.
Could you please let me know how should I solve this issue? I'm not even sure whether the problem lies with CrazySim or CrazySwarm2.

Do you know what might be causing this issue?

llanesc, 2024-11-08:
If you run too many drones your Gazebo real time factor will decrease below 70-80% which means you will probably lose packets and the drones become unstable. The amount of drones you can support may depend on your PC setup. How many are you running and what is your Gazebo real time factor? The startTrajectory() problem i'm not totally sure on, but I think you need to upload a trajectory before starting it.

ArkhamKnightNotRedHood, 2024-11-09:
> If you run too many drones your Gazebo real time factor will decrease below 70-80% which means you will probably lose packets and the drones become unstable. The amount of drones you can support may [...]

I'm sure that's not the reason. In fact I‘m just running the Crazyfile_examples code like nice_hover and figure8. I only run 8 drones and also try to decrease to 1 drones, but goto() still run wrong. I indeed upload a trajectory before starting it but it can not move at all.

llanesc, 2024-11-11:
> I'm sure that's not the reason. In fact I‘m just running the Crazyfile_examples code like nice_hover and figure8. I only run 8 drones and also try to decrease to 1 drones, but goto() still run [...]

I haven't seen this behavior before. Could you provide some screenshots or a video demonstrating the simulation and your issue?

ArkhamKnightNotRedHood, 2024-11-13:
> I haven't seen this behavior before. Could you provide some screenshots or a video demonstrating the simulation and your issue?

Thank you for your previous response. I have solved the issues of goto(), you are right, sorry for my careless before. But start_trajectory() can still not work. I read the crazyflie.py and found that upload_trajectory() does not exist, and the same applies to cmdposition(), notifySetpointsStop(), getparam(), and so on. It looks like the version of crazyswarm2 in the CrazySim has not been updated. I am trying to control it using low-level instructions.

llanesc, 2024-11-14:
Feel free to use the up-to-date version of Crazyswarm2. Mine was specifically for the MPC demo. I'm going to update the firmware and cflib repos soon since they've been out of date for a few months.

ArkhamKnightNotRedHood, 2024-11-15:
> Feel free to use the up-to-date version of Crazyswarm2. Mine was specifically for the MPC demo. I'm going to update the firmware and cflib repos soon since they've been out of date for a few months.

Excuse me, there are new issues happened, I find that the cmdFullState() can not move the drones. As I know, cmdFullState() is implemented  by the ROS2 topic publisher. After debugging, I confirmed that cmdFullStateMsg is indeed continuously updating position, speed, and other information, but when executing self.cmdFullStatePublisher.publish(self.cmdFullStateMsg), the drone does not move in the gazebo. I checked the MPC module you specifically use, and I found that movement is also achieved through the publisher. So why is this issue occurring here?

llanesc, 2024-11-15:
The MPC module uses a custom attitude setpoint publisher with Crazyswarm2. I haven't used cmdFullState before, but you can look at the [cflib example](https://github.com/bitcraze/crazyflie-lib-python/blob/master/examples/autonomy/full_state_setpoint_demo.py) for a demo on how to use it.

closed 2024-12-02 by llanesc