gtfactslab/CrazySim #20: Cross platform building

r-bt, 2025-02-16:
Hi I noticed in the README it mentioned

```
SITL integration with Kbuild is being developed for cross-platform building.
```

I'm interesting in getting CrazySim building on macOS and was wondering if there is any update with this?

Thanks

llanesc, 2025-02-16:
I've been really busy with my PhD lately and haven't had the time to fully set it up. I remember getting the Crazyflie binaries to compile with just Kbuild, but CMake is still necessary for building the Gazebo plugin and setting up dependencies. What do you need to compile on macOS?

r-bt, 2025-02-16:
No worries of course. I unfortunately don't have a Linux machine do I was trying to compile the gaezbo plugin to run the simulations on my Mac. I wanted to check was there any branches for this, etc before looking into in depth myself

llanesc, 2025-02-16:
They were local branches on my machine from a few months ago, but they're from an older master version. I'll make a new branch in the next few days with the Crazyflie firmware KBuild version.

llanesc, 2026-02-11:
@j1mb0o i had claude setup a kbuild version i can push soon to a dev-kbuild branch soon. however, the gazebo building is with cmake though. i'm not sure if it works for mac either. @r-bt did have a pull request https://github.com/llanesc/crazyflie-firmware/pull/2 for kbuild for macos but i don't have one to test on. perhaps you can try that.

j1mb0o, 2026-02-13:
@llanesc so instead of 
```bash
mkdir -p sitl_make/build && cd $_
cmake ..
make all
``` 
i should run the commands you gave above?