gtfactslab/CrazySim #23: Cfclient stuck with "Connecting to udp://0.0.0.0:19850"

joe-li-cal, 2025-04-09:
Hi all!

I am trying to run crazysim and I have made sure to clone and install both customized versions of cflib and cfclient. The process is mostly smooth and cfclient is able to scan and find the correct address which is `udp://0.0.0.0:19850`. 

However, when I press "connect", both the simulation and the client become stuck. Nothing happens after and the connection never happens. 

The last log for gazebo is: `[Msg] Received CfLib handshake message...`

And the log for cfclient is:
```
INFO:cfclient.utils.input:Using device blacklist [(VirtualBox|VMware)]
INFO:cfclient.ui.tab_toolbox:No config found for toolbox_areas
INFO:cfclient.ui.tab_toolbox:No config found for toolbox_areas
INFO:cfclient.ui.tab_toolbox:No config found for toolbox_areas
INFO:cfclient.ui.tab_toolbox:No config found for toolbox_areas
INFO:cfclient.ui.tab_toolbox:No config found for toolbox_areas
INFO:cfclient.ui.tab_toolbox:No config found for toolbox_areas
INFO:cflib.crtp.radiodriver:v99.55 dongle with serial N/A found
INFO:cfclient.ui.tab_toolbox:No config found for toolbox_areas
INFO:cfclient.ui.tab_toolbox:No config found for toolbox_areas
INFO:cfclient.ui.tab_toolbox:No config found for toolbox_areas
INFO:cfclient.ui.tab_toolbox:No config found for toolbox_areas
INFO:cfclient.ui.tab_toolbox:No config found for open_toolboxes
INFO:cfclient.ui.tab_toolbox:No config found for open_toolboxes
INFO:cfclient.utils.input:Selected MUX: Normal
INFO:cfclient.ui.main:Role of Microsoft X-Box 360 pad is Device
INFO:cfclient.utils.input.mux:Adding device Microsoft X-Box 360 pad to MUX Normal
INFO:cflib.drivers.cfusb:Looking for devices....
INFO:cflib.crtp.udpdriver:Initialized UDP driver.
INFO:cflib.crtp.prrtdriver:Initialized PRRT driver.
INFO:cflib.crtp.radiodriver:v99.55 dongle with serial N/A found
INFO:cflib.drivers.cfusb:Looking for devices....
INFO:cflib.crtp.udpdriver:Initialized UDP driver.
INFO:cflib.crtp.prrtdriver:Initialized PRRT driver.
INFO:cflib.crazyflie:Callback->Connection initialized[udp://0.0.0.0:19850]
INFO:cflib.crtp.udpdriver:Initialized UDP driver.
INFO:cflib.crazyflie:We are connected[udp://0.0.0.0:19850], request connection setup
INFO:cflib.crazyflie.platformservice:Request _request_protocol_version()
```

Could anyone please provide some insights? Thank you so much!

llanesc, 2025-04-10:
Can you confirm that you are using the sitl-release branch of https://github.com/llanesc/crazyflie-clients-python? You may have to uninstall the main one first if you already had that installed.

joe-li-cal, 2025-04-11:
Thank you so much for your response! Yes, I can confirm that I am using the sitl-release branch for cfclient. I suspect that cfclient might not be the problem because it successfully scans and finds the udp address but gets hang on "connection setup". I reinstalled everything one more time and I still get the same error message in my cfclient terminal:

`INFO:cflib.crtp.prrtdriver:Initialized PRRT driver.
INFO:cflib.crazyflie:Callback->Connection initialized[udp://0.0.0.0:19850]
NFO:cflib.crtp.udpdriver:Initialized UDP driver.
INFO:cflib.crazyflie:We are connected[udp://0.0.0.0:19850], request connection setup
INFO:cflib.crazyflie.platformservice:Request _request_protocol_version()
`
it seems that there is something wrong with the cflib package?

afonsob5, 2025-05-31:
Hi, I'm with the same problem! Have you fixed it?

VitinLima, 2025-06-25:
Hello, I'm having the same issue here, stuck on cfclient connecting.

llanesc, 2025-06-25:
> Hello, I'm having the same issue here, stuck on cfclient connecting. Could you share which version of cflib are you working with?

I am using the cflib library provided in this repo.

VitinLima, 2025-06-29:
Hi, I've been able to make it work... on another more powerful machine... Let me elaborate:
I was working on my personal computer, since at this stage of my internship I was not supposed to run any physics simulations or anything too heavy. My personal machine is quite old (about 9yo) but it serves me well. My colleague at the internship cloned the repo, followed the instructions and voilà, it worked right out of the box. However, the only difference I could tell from our builds was the fact that she ran on the labs machine, which is much more powerful.
Also, I've been able to run the ros2 example in my own pc... so the only thing that wouldn't work is the cfclient connecting with udp...
I resume, I guess the workload of the gazebo simulation is too much for some machines and it just takes forever for cfclient to connect, if it ever does, or maybe its just some timing issue that doesn't appear when the simulation runs more smoothly. That is my theory anyways... But now it is running well on my labs computer, so there's that. If anyone else having the same problem tries to run on a faster computer and that solves, let us know here pls.

whui80327-cell, 2025-11-25:
Hello all! When I connect to crazyflie using cfclient, the connection is successful. However, when I press 'takeoff', the cfclient page does not respond, and after a while it prompts me to either wait or force quit. Could anyone provide some insights? Thank you.

Oculux314, 2025-11-25:
Hello, I'm facing the same issue - when attempting to connect with cfclient it just hangs.

```
> bash sitl_singleagent.sh -m crazyflie -x 0 -y 0

killing running crazyflie firmware instances
Starting gazebo
[Msg] Gazebo Sim Server v7.9.0
[Msg] Loading SDF world file[/home/anyone/Documents/nwil508/CrazySim/crazyflie-firmware/tools/crazyflie-simulation/simulator_files/gazebo/launch/../../../../../tools/crazyflie-simulation/simulator_files/gazebo/worlds/crazysim_default.sdf].
[Msg] Serving entity system service on [/entity/system/add]
[Msg] Create service on [/world/crazysim_default/create]
[Msg] Remove service on [/world/crazysim_default/remove]
[Msg] Pose service on [/world/crazysim_default/set_pose]
[Msg] Pose service on [/world/crazysim_default/set_pose_vector]
[Msg] Light configuration service on [/world/crazysim_default/light_config]
[Msg] Physics service on [/world/crazysim_default/set_physics]
[Msg] SphericalCoordinates service on [/world/crazysim_default/set_spherical_coordinates]
[Msg] Enable collision service on [/world/crazysim_default/enable_collision]
[Msg] Disable collision service on [/world/crazysim_default/disable_collision]
[Msg] Material service on [/world/crazysim_default/visual_config]
[Msg] Material service on [/world/crazysim_default/wheel_slip]
[Msg] Loaded level [default]
[Msg] Serving world controls on [/world/crazysim_default/control], [/world/crazysim_default/control/state] and [/world/crazysim_default/playback/control]
[Msg] Serving GUI information on [/world/crazysim_default/gui/info]
[Msg] World [crazysim_default] initialized with [default_physics] physics profile.
[Msg] Serving world SDF generation service on [/world/crazysim_default/generate_world_sdf]
[Msg] Serving world names on [/gazebo/worlds]
[Msg] Resource path add service on [/gazebo/resource_paths/add].
[Msg] Resource path get service on [/gazebo/resource_paths/get].
[Msg] Resource path resolve service on [/gazebo/resource_paths/resolve].
[Msg] Resource paths published on [/gazebo/resource_paths].
[Msg] Server control service on [/server_control].
[Msg] Found no publishers on /stats, adding root stats topic
[Msg] Found no publishers on /clock, adding root clock topic
[Msg] Serving scene information on [/world/crazysim_default/scene/info]
[Msg] Serving graph information on [/world/crazysim_default/scene/graph]
[Msg] Serving full state on [/world/crazysim_default/state]
[Msg] Serving full state (async) on [/world/crazysim_default/state_async]
[Msg] Publishing scene information on [/world/crazysim_default/scene/info]
[Msg] Publishing entity deletions on [/world/crazysim_default/scene/deletion]
[Msg] Publishing state changes on [/world/crazysim_default/state]
[Msg] Publishing pose messages on [/world/crazysim_default/pose/info]
[Msg] Publishing dynamic pose messages on [/world/crazysim_default/dynamic_pose/info]
/home/anyone/Documents/nwil508/CrazySim/crazyflie-firmware/tools/crazyflie-simulation/simulator_files/gazebo/launch/../../../../../tools/crazyflie-simulation/simulator_files/gazebo/models/crazyflie/model.sdf.jinja -> /tmp/crazyflie_0.sdf
Spawning crazyflie_0 at 0 0
data: true

starting instance 0 in /home/anyone/Documents/nwil508/CrazySim/crazyflie-firmware/sitl_make/build/0
Starting gazebo gui
[Msg] OdometryPublisher publishing odometry on [/cf_0/odom]
[Msg] OdometryPublisher publishing odometry with covariance on [/model/crazyflie_0/odometry_with_covariance]
[Msg] OdometryPublisher publishing Pose_V (TF) on [/model/crazyflie_0/pose]
[Msg] Init subs and Pubs done : 
[Msg] Received firmware handshake message...
[Wrn] [Component.hh:144] Trying to serialize component with data type [N3sdf3v135WorldE], which doesn't have `operator<<`. Component will not be serialized.
[Wrn] [Model.hh:69] Skipping serialization / deserialization for models with //pose/@relative_to attribute.
libEGL warning: egl: failed to create dri2 screen
libEGL warning: egl: failed to create dri2 screen
libEGL warning: egl: failed to create dri2 screen
libEGL warning: egl: failed to create dri2 screen
[Msg] Received CfLib handshake message...
```

```
> cfclient

INFO:cfclient.gui:Disabling STL printouts
INFO:cfclient.utils.input.inputreaders:Input readers: ['linuxjsdev', 'pysdl2']
INFO:cfclient.utils.input.inputreaders:Successfully initialized [linuxjsdev]
INFO:cfclient.utils.input.inputreaders:Could not initialize [pysdl2]: No SDL2 support on Linux
[... 9 lines cut ...]
INFO:cfclient.utils.input:Using device blacklist [(VirtualBox|VMware)]
INFO:cfclient.ui.tab_toolbox:No config found for toolbox_areas
Cannot find a Crazyradio Dongle
INFO:cflib.drivers.cfusb:Looking for devices....
INFO:cflib.crtp.udpdriver:Initialized UDP driver.
INFO:cflib.crtp.prrtdriver:Initialized PRRT driver.
INFO:cfclient.ui.tab_toolbox:No config found for toolbox_areas
INFO:cfclient.ui.tab_toolbox:No config found for toolbox_areas
INFO:cfclient.ui.tab_toolbox:No config found for toolbox_areas
INFO:cfclient.ui.tab_toolbox:No config found for toolbox_areas
INFO:cfclient.ui.tab_toolbox:No config found for toolbox_areas
INFO:cfclient.ui.tab_toolbox:No config found for toolbox_areas
INFO:cfclient.ui.tab_toolbox:No config found for toolbox_areas
INFO:cfclient.ui.tab_toolbox:No config found for toolbox_areas
INFO:cfclient.ui.tab_toolbox:No config found for toolbox_areas
INFO:cfclient.ui.tab_toolbox:No config found for open_toolboxes
INFO:cfclient.ui.tab_toolbox:No config found for open_toolboxes
WARNING:cfclient.ui.main:Cannot find a Crazyradio Dongle
Cannot find a Crazyradio Dongle
INFO:cflib.drivers.cfusb:Looking for devices....
INFO:cflib.crtp.udpdriver:Initialized UDP driver.
INFO:cflib.crtp.prrtdriver:Initialized PRRT driver.
```

https://github.com/user-attachments/assets/ddac4bfe-c935-432e-a6d3-c2b8ccebe12f

Oculux314, 2025-12-04:
It's really strange - I've played around with the library a bit more, in particular adding logging to the crazysim_plugin.cpp file. As part of this, I've ran `make all` multiple times. Almost randomly, sometimes the resultant binary works and sometimes it doesn't. I can only guess that there's some sort of race condition, or potentially something to do with the state of the computer since going away for a long time and coming back can cause a working binary to not work or vice versa. This sounds really weird - and I've never encountered a program which did this before.

For anyone else encountering this issue, the best I can suggest is to add logging around the "Received CfLib handshake message..." message in crazysim_plugin.cpp, and rerun `make all` plenty of times when making these changes (my best guess right now is that the binary is very sensitive to randomness, e.g. with a race condition, so simply remaking the binary plenty of times produces a working binary occasionally).

Edit: I should add I think it's the simulator, not the cflib library or cfclient, since both cflib and cfclient work or break in sync, and its the simulator C++ code I've been regenerating.