gtfactslab/CrazySim #27: Crazyflie object has no attribute 'link_statistics'

shaumikkalwit, 2025-06-11:
I am on Ubuntu 22, using ROS humble and Gazebo Garden.

When I try and run `ros2 launch crazyflie launch.py backend:=cflib` in my ROS 2 workspace on one terminal, and `bash tools/crazyflie-simulation/simulator_files/gazebo/launch/sitl_multiagent_square.sh -n 4 -m crazyflie` on one of the other terminals, a blank white NiceGUI window opens up (which would otherwise be visualizing whatever crazyswarm2 script I am running if not doing it with CrazySim), and the CrazySim drones don't do anything.

While this is happening, I get this message in the terminal running the crazyswarm2 backend:
`$ ros2 launch crazyflie launch.py backend:=cflib
[INFO] [launch]: All log files can be found below /home/optimalx/.ros/log/2025-06-11-13-17-33-775423-anemoi-27011
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [crazyflie_server.py-1]: process started with pid [27012]
[INFO] [teleop-2]: process started with pid [27014]
[INFO] [joy_node-3]: process started with pid [27016]
[INFO] [gui.py-4]: process started with pid [27018]
[teleop-2] [INFO] [1749665853.828254031] [teleop]: Mode changed to high_level
[crazyflie_server.py-1] Traceback (most recent call last):
[crazyflie_server.py-1]   File "/home/optimalx/drone_ws/install/crazyflie/lib/crazyflie/crazyflie_server.py", line 1280, in <module>
[crazyflie_server.py-1]     main()
[crazyflie_server.py-1]   File "/home/optimalx/drone_ws/install/crazyflie/lib/crazyflie/crazyflie_server.py", line 1271, in main
[crazyflie_server.py-1]     crazyflie_server = CrazyflieServer()
[crazyflie_server.py-1]   File "/home/optimalx/drone_ws/install/crazyflie/lib/crazyflie/crazyflie_server.py", line 165, in __init__
[crazyflie_server.py-1]     self.swarm._cfs[link_uri].cf.link_statistics.latency_updated.add_callback(partial(self._latency_callback, uri=link_uri))
[crazyflie_server.py-1] AttributeError: 'Crazyflie' object has no attribute 'link_statistics'
[ERROR] [crazyflie_server.py-1]: process has died [pid 27012, exit code 1, cmd '/home/optimalx/drone_ws/install/crazyflie/lib/crazyflie/crazyflie_server.py --ros-args -r __node:=crazyflie_server --params-file /tmp/launch_params_gm790whq --params-file /tmp/launch_params_orw75zvv'].`

I am not sure why this is happening, I tried deleting and reinstalling everything a few times. Thanks in advance for any help.

adbidwai, 2025-06-17:
@llanesc, any solutions or methods to debug the problem? Thanks!

llanesc, 2025-06-19:
> Still the same error unfortunately.
> 
> ```
> :~/drone_ws$ ros2 launch crazyflie launch.py backend:=cflib
> [INFO] [launch]: All log files can be found below /home/optimalx/.ros/log/2025-06-12-09-26-39-144773-anemoi-15697
> [INFO] [launch]: Default logging verbosity is set to INFO
> [INFO] [crazyflie_server.py-1]: process started with pid [15698]
> [INFO] [teleop-2]: process started with pid [15700]
> [INFO] [joy_node-3]: process started with pid [15702]
> [INFO] [gui.py-4]: process started with pid [15704]
> [teleop-2] [INFO] [1749738399.208238745] [teleop]: Mode changed to high_level
> [crazyflie_server.py-1] Traceback (most recent call last):
> [crazyflie_server.py-1]   File "/home/optimalx/drone_ws/install/crazyflie/lib/crazyflie/crazyflie_server.py", line 1280, in <module>
> [crazyflie_server.py-1]     main()
> [crazyflie_server.py-1]   File "/home/optimalx/drone_ws/install/crazyflie/lib/crazyflie/crazyflie_server.py", line 1271, in main
> [crazyflie_server.py-1]     crazyflie_server = CrazyflieServer()
> [crazyflie_server.py-1]   File "/home/optimalx/drone_ws/install/crazyflie/lib/crazyflie/crazyflie_server.py", line 165, in __init__
> [crazyflie_server.py-1]     self.swarm._cfs[link_uri].cf.link_statistics.latency_updated.add_callback(partial(self._latency_callback, uri=link_uri))
> [crazyflie_server.py-1] AttributeError: 'Crazyflie' object has no attribute 'link_statistics'
> [ERROR] [crazyflie_server.py-1]: process has died [pid 27772, exit code 1, cmd '/home/optimalx/drone_ws/install/crazyflie/lib/crazyflie/crazyflie_server.py --ros-args -r __node:=crazyflie_server --params-file /tmp/launch_params_ayneepo3 --params-file /tmp/launch_params_gjqg6vuk'].
> ```

Sorry about the late reply. I was on vacation the past week and haven't been able to check my emails. Are you sure you are using the sitl branch of my repo? That line shouldn't exist in that version.

shaumikkalwit, 2025-06-20:
> Sorry about the late reply. I was on vacation the past week and haven't been able to check my emails. Are you sure you are using the sitl branch of my repo? That line shouldn't exist in that version.

Hi! That worked, thank you! But I have a few follow up questions now. When I run the crazyswarm2 command now, I get 
```
$ ros2 launch crazyflie launch.py backend:=cflib
[INFO] [launch]: All log files can be found below /home/optimalx/.ros/log/2025-06-20-09-54-44-709981-anemoi-19705
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [motion_capture_tracking_node-1]: process started with pid [19706]
[INFO] [teleop-2]: process started with pid [19708]
[INFO] [joy_node-3]: process started with pid [19710]
[INFO] [crazyflie_server.py-4]: process started with pid [19712]
[INFO] [gui.py-5]: process started with pid [19714]
[motion_capture_tracking_node-1] terminate called after throwing an instance of 'rclcpp::exceptions::InvalidParameterValueException'
[motion_capture_tracking_node-1]   what():  No parameter value set
[teleop-2] [ERROR] [1750431284.765478335] [teleop]: Unknown mode none
[crazyflie_server.py-4] [INFO] [1750431285.669558471] [crazyflie_server]: [cf_3] is fully connected!
[crazyflie_server.py-4] [INFO] [1750431285.706017055] [crazyflie_server]: [cf_4] is fully connected!
[crazyflie_server.py-4] [INFO] [1750431285.706144311] [crazyflie_server]: [cf_2] is fully connected!
[crazyflie_server.py-4] [INFO] [1750431285.706269745] [crazyflie_server]: [cf_1] is fully connected!
[crazyflie_server.py-4] [INFO] [1750431285.706525240] [crazyflie_server]: All Crazyflies are fully connected!
[crazyflie_server.py-4] [INFO] [1750431285.706846076] [crazyflie_server]: [cf_1] commander.enHighLevel is set to 1
[crazyflie_server.py-4] [INFO] [1750431285.720659187] [crazyflie_server]: [cf_1] locSrv.extPosStdDev is set to 0.001
[crazyflie_server.py-4] [INFO] [1750431285.721045604] [crazyflie_server]: [cf_1] locSrv.extQuatStdDev is set to 0.05
[crazyflie_server.py-4] [INFO] [1750431285.739172453] [crazyflie_server]: [cf_1] stabilizer.controller is set to 2
[crazyflie_server.py-4] [INFO] [1750431285.739554704] [crazyflie_server]: [cf_1] stabilizer.estimator is set to 2
[crazyflie_server.py-4] [INFO] [1750431285.746674148] [crazyflie_server]: [cf_2] commander.enHighLevel is set to 1
[crazyflie_server.py-4] [INFO] [1750431285.757104610] [crazyflie_server]: [cf_2] locSrv.extPosStdDev is set to 0.001
[crazyflie_server.py-4] [INFO] [1750431285.757491047] [crazyflie_server]: [cf_2] locSrv.extQuatStdDev is set to 0.05
[crazyflie_server.py-4] [INFO] [1750431285.767342104] [crazyflie_server]: [cf_2] stabilizer.controller is set to 2
[crazyflie_server.py-4] [INFO] [1750431285.767713248] [crazyflie_server]: [cf_2] stabilizer.estimator is set to 2
[crazyflie_server.py-4] [INFO] [1750431285.773566195] [crazyflie_server]: [cf_3] commander.enHighLevel is set to 1
[crazyflie_server.py-4] [INFO] [1750431285.780917558] [crazyflie_server]: [cf_3] locSrv.extPosStdDev is set to 0.001
[crazyflie_server.py-4] [INFO] [1750431285.781265355] [crazyflie_server]: [cf_3] locSrv.extQuatStdDev is set to 0.05
[crazyflie_server.py-4] [INFO] [1750431285.786733297] [crazyflie_server]: [cf_3] stabilizer.controller is set to 2
[crazyflie_server.py-4] [INFO] [1750431285.787053762] [crazyflie_server]: [cf_3] stabilizer.estimator is set to 2
[crazyflie_server.py-4] [INFO] [1750431285.790442340] [crazyflie_server]: [cf_4] commander.enHighLevel is set to 1
[crazyflie_server.py-4] [INFO] [1750431285.798299595] [crazyflie_server]: [cf_4] locSrv.extPosStdDev is set to 0.001
[crazyflie_server.py-4] [INFO] [1750431285.798651709] [crazyflie_server]: [cf_4] locSrv.extQuatStdDev is set to 0.05
[crazyflie_server.py-4] [INFO] [1750431285.804896612] [crazyflie_server]: [cf_4] stabilizer.controller is set to 2
[crazyflie_server.py-4] [INFO] [1750431285.805222015] [crazyflie_server]: [cf_4] stabilizer.estimator is set to 2
[crazyflie_server.py-4] [INFO] [1750431285.809411160] [crazyflie_server]: All Crazyflies parameters are initialized.
[crazyflie_server.py-4] [INFO] [1750431285.809915159] [crazyflie_server]: [cf_1] setup logging for pose at freq 10
[crazyflie_server.py-4] Unable to find variable [pm.vbatMV]
[crazyflie_server.py-4] pm.vbatMV not in TOC, this block cannot be used!
[crazyflie_server.py-4] [ERROR] [1750431285.810496417] [crazyflie_server]: [cf_1] Could not start log configuration,'Variable pm.vbatMV not in TOC' not found in TOC
[crazyflie_server.py-4] [INFO] [1750431285.812952063] [crazyflie_server]: [cf_2] setup logging for pose at freq 10
[crazyflie_server.py-4] Unable to find variable [pm.vbatMV]
[crazyflie_server.py-4] pm.vbatMV not in TOC, this block cannot be used!
[crazyflie_server.py-4] [ERROR] [1750431285.813359062] [crazyflie_server]: [cf_2] Could not start log configuration,'Variable pm.vbatMV not in TOC' not found in TOC
[crazyflie_server.py-4] [INFO] [1750431285.814806800] [crazyflie_server]: [cf_3] setup logging for pose at freq 10
[crazyflie_server.py-4] Unable to find variable [pm.vbatMV]
[crazyflie_server.py-4] pm.vbatMV not in TOC, this block cannot be used!
[crazyflie_server.py-4] [ERROR] [1750431285.815127936] [crazyflie_server]: [cf_3] Could not start log configuration,'Variable pm.vbatMV not in TOC' not found in TOC
[crazyflie_server.py-4] [INFO] [1750431285.816665923] [crazyflie_server]: [cf_4] setup logging for pose at freq 10
[crazyflie_server.py-4] Unable to find variable [pm.vbatMV]
[... cut ...]
[crazyflie_server.py-4] [ERROR] [1750431285.810496417] [crazyflie_server]: [cf_1] Could not start log configuration,'Variable pm.vbatMV not in TOC' not found in TOC
[crazyflie_server.py-4] [INFO] [1750431285.812952063] [crazyflie_server]: [cf_2] setup logging for pose at freq 10
```

Along with this, it also creates two files in the root directory of my ROS 2 workspace named "tmp_motion_capture.yaml" and "tmp_server.yaml", which I'm guessing it creates because it could not find it initially.

I am not sure why that's happening and was wondering if you could help me out.